ci-group/revolve2 #17: Profile performance improvements

surgura, 2021-12-02:
Profile and make a list of performance improvements. Create issues accordingly.

surgura, 2022-03-23:
besides checking the optimizer part, rpi controller needs profiling because it is very slow

surgura, 2023-09-07:
This should be the last thing we do before releasing version 1.

oliverweissl, 2023-11-20:
Im going to profile all examples using cProfile + SnakeViz

oliverweissl, 2023-11-20:
from preliminary performance tracking the following issues arise:

Performance issues:

simulation/../_multi_body_system:calculate_aabb -> 10-18% of runtime
modular_robot/../_body.py: grid_position -> 2.5% of runtime for ea_db example.

-> rng Generator from numpy also takes significant time in most runs. -> mainly tournament selection.
-> session commits are time consuming. 
-> _find_and_load takes a lot (loading modules etc)

oliverweissl, 2023-11-20:
-> Quaternions could be replaced with rotation matrices --> quat to vec3 calculations take very long

-> calculate_aabb has been improved from O(n³) to O(n) in #370

surgura, 2023-11-23:
so are those performance issues relevant when compared to how long simulation takes?

oliverweissl, 2023-12-15:
This can be picked up after 1.0.0
--> possible further improvements move to rotation matrices ( not that significant )