ci-group/revolve2 #100: Pareto-dominance fitness function

karinemiras, 2022-04-06:


surgura, 2022-04-07:
Can you describe what the requirements would be for such a function(or set or functions?) Maybe provide links to algorithms

karinemiras, 2022-04-07:
It is simple: the function should calculate the number of dominated solutions of a solution (e.g., for each robot against all robots in the pop). Dominating means that it is not worse in any of the objectives and better in at least one. The number of objectives should be a param (as many as the user provides).

surgura, 2022-04-07:
Ok so given an evaluated individual and an evaluated population, count the number of individuals in the population that are worse than the given individual, provided some user function over fitness.

oliverweissl, 2023-11-22:
included in #381

closed 2023-11-22 by oliverweissl