ci-group/revolve2 #103: Plasticoding

karinemiras, 2022-04-06:


surgura, 2022-04-07:
It is important to keep in mind that before this can merged into Revolve it must be some kind of 'generic' framework.

surgura, 2022-04-07:
If you are working on this in a personal repository it would we nice if you could post a link here so we can track that, if possible/desired

karinemiras, 2022-04-07:
I'm not sure what you mean by framework. Plasticoding is a representation, with peculiar capabilities.

surgura, 2022-04-07:
If it is in revolve it should be generic and theoretically useful for people outside of our research group. Basically what I mean is it should not simply be a set of hardcoded rules that only work for one specific experiment

bobykhani, 2022-04-07:
> If you are working on this in a personal repository it would we nice if you could post a link here so we can track that, if possible/desired

[lsystem-repo](https://github.com/bobykhani/revolve2/tree/lsystem)

This is the link to the branch that I'm working on it. It has an L-system body + random CPG brain,
I'm about to finish CPPN-CPG with multi-neat for brain genome and commit it but still need some fixation to merge with the master branch.

karinemiras, 2022-04-07:
Environmental regulation is relevant for the whole community of artificial
life. The rules are not hard coded, but evolvable.

karinemiras, 2022-04-07:
I think before we think of making things useful to the community we should make it useful for our group, which is challenging enough for now. Specially because in the end, each lab uses their own system. In any case, plasticity could theoretically be implemented with any representation, but making it work is not particularly trivial. And this is why I don't think it's feasible yet to have a "framework". It's not simply a question of implementing the dynamics of it. That would be purely a coding challenge. The real challenge is composing a design that, given the rest of the system (search space, task, environment, etc) , allows the emergence of the right conditions for plasticity to be selected for. I could not make it work with cppns yet. Only with the lsystem. But I'd love to have it for both. Note that regulation is inherently part of the decoding, during which the "choice" of genetic expression is done, mediated by the environment. Some consider the decoding as part  of the representation and some don't. In any case, they are coupled together, and the decoding depends on the encoding. This way, one could devise  multiple regulatory designs for a same representation, but all of them would depend on the nature of this representation. More than that, it would be hard to take, for instance, a regulatory design of an lsystem and plug into a cppn (just like a mutation operator wouldn't fit).