ci-group/revolve2 #104: Robot pictures: either top-down unactuated-simulated-robot or 2D drawing

karinemiras, 2022-04-06:


surgura, 2022-04-07:
As a workaround there is currently rerunning of the robot. Extendign that to create images from desired poses and angles would be nice.

karinemiras, 2022-04-07:
Yes. having to simulate each robot manually is not practical just to see the shape. I look at hundreds of pictures during my analysis. It has to be automatic. This is why I made the 2D visualization in the past. If taking pictures automatically is difficult maybe we can implement the 2d drawings again for now. Sayif converted it to python: https://github.com/ci-group/revolve/blob/plasticoding/pyrevolve/revolve_bot/render/render.py

surgura, 2022-04-07:
Yes I was thinking of rendering functionality indeed

oliverweissl, 2023-11-24:
worked on in #384

closed 2023-11-24 by oliverweissl