ci-group/revolve2 #97: morphological measures

karinemiras, 2022-04-06:


surgura, 2022-04-07:
Can you elaborate on what this means?

karinemiras, 2022-04-07:
These are descriptors of body traits, and should be identical to the ones in Revolve 1, available here: https://github.com/ci-group/revolve/blob/plasticoding/pyrevolve/revolve_bot/measure/measure_body.py

surgura, 2022-04-07:
A start on this has already been made with the analyzer: https://github.com/ci-group/revolve2/blob/master/core/revolve2/core/modular_robot/_analyzer.py
#108

karinemiras, 2022-04-07:
Oh, so the concept of analyzer has changed? Before it had to do with collisions or so (I think).

surgura, 2022-04-20:
Modular robot body has been refactored to make adding measures easier. Actually measures can be added by the people who need them on a demand basis and specific issues for each measure can be made

closed 2022-04-20 by surgura