ci-group/revolve2 #99: Novelty search

karinemiras, 2022-04-06:


surgura, 2022-04-07:
Is this a specific algorithm? Or does this simply mean selection based on phenotypical features? In the latter case we would simply need to provide the ability to measure modular robots(#97 )? The existing EA optimizer might alredy provide a framework for the evolution process.

karinemiras, 2022-04-07:
Neither one nor another, but a hybrid. This is an optimization towards diversity: the dimensions should be a parameter - the user can choose which traits (behavioral, phenotypical, genotypical, etc) to use for defining diversity. It requires a measure of diversity, but also an archive to make sure the search won't go in circles.
You can read the original paper: https://ieeexplore.ieee.org/document/6793380
Or take a look at the current code: https://github.com/ci-group/revolve/blob/plasticoding/pyrevolve/evolution/population.py
relevant methods of the file script: def calculate_novelty, def calculate_distances, def collect_measures, def update_archive

surgura, 2022-04-07:
Let's create a new example and meanwhile implement features in revolve we need for that