ci-group/revolve2 #203: concurrent.futures fails on Windows

tckarenchiang, 2023-01-11:
There are students who run `revolves2` on Windows. And, they have the following error which I reproduced in the below. It seems that `concurrent.futures` fails on Windows because it lacks `fork()` system. I am not entirely sure how to fix this, @surgura .
```
(.venv) C:\Users\ttingchia\Desktop\revolve2-0.3.6-beta1\revolve2-0.3.6-beta1\examples>python simulate_mujoco/main.py
concurrent.futures.process._RemoteTraceback:
"""
Traceback (most recent call last):
  File "C:\Users\ttingchia\AppData\Local\Programs\Python\Python38\lib\concurrent\futures\process.py", line 239, in _process_worker
    r = call_item.fn(*call_item.args, **call_item.kwargs)
  File "C:\Users\ttingchia\Desktop\.venv\lib\site-packages\revolve2\runners\mujoco\_local_runner.py", line 92, in _run_environment
    model = mujoco.MjModel.from_xml_string(cls._make_mjcf(env_descr))
  File "C:\Users\ttingchia\Desktop\.venv\lib\site-packages\revolve2\runners\mujoco\_local_runner.py", line 287, in _make_mjcf
    mujoco.mj_saveLastXML(botfile.name, model)
mujoco.FatalError: File not found
"""

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "simulate_mujoco/main.py", line 114, in <module>
    asyncio.run(main())
  File "C:\Users\ttingchia\AppData\Local\Programs\Python\Python38\lib\asyncio\runners.py", line 43, in run
    return loop.run_until_complete(main)
  File "C:\Users\ttingchia\AppData\Local\Programs\Python\Python38\lib\asyncio\base_events.py", line 608, in run_until_complete
    return future.result()
  File "simulate_mujoco/main.py", line 108, in main
    await sim.simulate(robot, 60)
  File "simulate_mujoco/main.py", line 87, in simulate
    await runner.run_batch(batch)
  File "C:\Users\ttingchia\Desktop\.venv\lib\site-packages\revolve2\runners\mujoco\_local_runner.py", line 239, in run_batch
    results = BatchResults([future.result() for future in futures])
  File "C:\Users\ttingchia\Desktop\.venv\lib\site-packages\revolve2\runners\mujoco\_local_runner.py", line 239, in <listcomp>
    results = BatchResults([future.result() for future in futures])
  File "C:\Users\ttingchia\AppData\Local\Programs\Python\Python38\lib\concurrent\futures\_base.py", line 439, in result
    return self.__get_result()
  File "C:\Users\ttingchia\AppData\Local\Programs\Python\Python38\lib\concurrent\futures\_base.py", line 388, in __get_result
    raise self._exception
mujoco.FatalError: File not found
```

surgura, 2023-01-12:
I don't think this has to do with fork or concurrent. saveLastXML seems to fail. You can check which temporary file it tries to read from and write to. Big chance this is in /tmp, which does not exist on Windows.

tckarenchiang, 2023-01-15:
> I don't think this has to do with fork or concurrent. saveLastXML seems to fail. You can check which temporary file it tries to read from and write to. Big chance this is in /tmp, which does not exist on Windows.

You are right, it's not about fork or concurrent. But, the `/tmp` does exist on Windows, and it's usually located in `%userprofile%\AppData\Local\Temp`. The problem is that on Windows, the tempfile.name can't be opened a second time when the named temporary file is still open (when `delete=True`, by default), which means the mujoco model can't be saved as a `_mujoco.urdf` file when the file is still open. See [tempfile.NamedTemporaryFile](https://docs.python.org/3/library/tempfile.html#tempfile.NamedTemporaryFile)

> Whether the name can be used to open the file a second time, while the named temporary file is still open, varies across platforms (it can be so used on Unix; it cannot on Windows). 

A workaround I believe would be setting `delete=False` (so that `FILE_SHARE_DELETE` flag won't be set, see a related issue: [NamedTemporaryFile opened twice on Windows](https://github.com/python/cpython/issues/88221)) and manually delete the temporary file. This way, it will work on both Linux and Windows.

PR #204 (2023-01-19): fix #203: set delete parameter to False so that the temp file can be opened a second time on Windows
As per subject, this fixes #203. 

I've tested it on both Linux and Windows.
commit b024e101 (2023-01-20): Fix issue #203 where tempfile.NamedTemporaryFile fails on Windows

* fix #203: set delete parameter to False so that the temp file can be open a second time on Windows

And, manually delete the temp file once the mujoco model is saved

* Add an error catching in case the mujuco model saving fails,

and make sure temp file is always deleted

* Set delete param to False only when the first attempt (delete=True) fails.

This way, we minimize the chance of failing to delete the temp file for Linux users

* Fix a typo
closed 2023-01-20 by tckarenchiang