ci-group/revolve2 #218: [WARNING] [core] Nan, Inf or huge value in QVEL at DOF 0. The simulation is unstable. Time = 0.0280.

dangbert, 2023-04-18:
[image]

Running into this error when running `examples/optimize_modular/optimize.py`  On my machine (Ubuntu 22.04, Python 3.8.10), the code hangs after this warning (it hung over night and never got past the warning).  If i hit `<ctrl>c` it then continues to the next environment and hits the error again and hangs.

I ran `git bisect` to investigate and I believe this problem started with this commit de0c322f93c83

[image]

And indeed if I `git checkout e578718b03` (the parent of this commit), then I can run the example code without this problem:
[image]

oliverweissl, 2023-04-18:
This might be due to the `timestep` variable in `_local_runner.py` -> line 252.
Decrease it and it should be stable again

surgura, 2023-04-18:
@tckarenchiang let's make this a public setting, because it depends on the specific robot that is being simulated.

surgura, 2023-04-28:
Resolved in #219 , will be in next release