ci-group/revolve2 #227: New tutorials

surgura, 2023-05-15:
The old tutorials(in the documentation, not the examples) are/will be soon wildly outdated.

surgura, 2023-08-30:
For beta2 release we will use the new examples instead of writing tutorials. I think that is good enough for now. We want the release rather sooner than later.

surgura, 2023-10-23:
@oliverweissl I think the examples + the current installation and getting started guide are sufficient. What do you think? I think we can close this issue.

oliverweissl, 2023-10-23:
yes i think they are sufficient aswell, besides the RPI part which is WIP

surgura, 2023-10-23:
yeah for all new features we should add new examples. and we can keep adding examples every time there are questions about things

closed 2023-10-23 by surgura