ci-group/revolve2 #241: ValueError when using LocalRunner through rerun_best.py or optimize.py

oliverweissl, 2023-06-23:
The following error is thrown:
`ValueError: Error: mass and inertia of moving bodies must be larger than mjMINVAL
Object name = robot_0/, id = 1, line = 15, column = -1`

- python version: 3.10.6
- occurs in revolve2 releases: [v0.3.9-beta1](https://github.com/ci-group/revolve2/releases/tag/v0.3.9-beta1) & [v0.3.10-beta1](https://github.com/ci-group/revolve2/releases/tag/v0.3.10-beta1)

The error gets thrown when running the `optimize.py` given in the examples folder. Additionally when running `rerun_best.py`, the same error gets thrown.

The databases used for  `rerun_best.py` were generated in a different python environment using python 3.9.9.

oliverweissl, 2023-06-25:
can be fixed by using `mujoco==2.3.5` in `runners/mujoco/setup.py `

surgura, 2023-06-26:
Reopening because it occurs with the latest Mujoco version. Should investigate and solve before next release.

surgura, 2023-08-14:
Appears solved with mujoco 2.3.7, but will investigate what broke in 2.3.6 because it might be a bug on our end.

surgura, 2023-08-17:
Significant digging yields the conclusion that a bug was introduced in the Mujoco XML writer in version 2.3.6, which was again fixed in version 2.3.7. Our code did not cause this bug. We safely ignore and make sure we don't use version 2.3.6.

closed 2023-08-17 by surgura