ci-group/revolve2 #242: Use official Mujoco viewer

surgura, 2023-08-21:
It appears deepminds python dev team has released python bindings for the Mujoco viewer. That means we can drop the use of https://github.com/rohanpsingh/mujoco-python-viewer and move to the official viewer.

surgura, 2023-09-07:
I tested it and it definitely works to run and render the robot using the built-in mujoco viewer. However, the 3rd party software we are using now supports things like pausing and speeding up out of the box and that's very convenient. So, sure we have to drop that package at some time in the future, but for now it's not worth the effort. putting this issue in the backlog.

oliverweissl, 2024-04-30:
- Native viewer support added

closed 2024-04-30 by oliverweissl