ci-group/revolve2 #421: Mac install not working

andricmitrovic, 2024-01-15:
Building wheel for RPi.GPIO (setup.py) ... error
  error: subprocess-exited-with-error
  
  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [30 lines of output]
      running bdist_wheel
      running build
      running build_py
      creating build
      creating build/lib.macosx-13.3-arm64-cpython-311
      creating build/lib.macosx-13.3-arm64-cpython-311/RPi
      copying RPi/__init__.py -> build/lib.macosx-13.3-arm64-cpython-311/RPi
      creating build/lib.macosx-13.3-arm64-cpython-311/RPi/GPIO
      copying RPi/GPIO/__init__.py -> build/lib.macosx-13.3-arm64-cpython-311/RPi/GPIO
      running build_ext
      building 'RPi._GPIO' extension
      creating build/temp.macosx-13.3-arm64-cpython-311
      creating build/temp.macosx-13.3-arm64-cpython-311/source
      clang -Wsign-compare -Wunreachable-code -DNDEBUG -g -fwrapv -O3 -Wall -stdlib=libc++ -I/Users/nikolaandricmitrovic/PycharmProjects/revolve2/.venv/include -I/Users/nikolaandricmitrovic/.pyenv/versions/3.11.6/include/python3.11 -c source/c_gpio.c -o build/temp.macosx-13.3-arm64-cpython-311/source/c_gpio.o
      source/c_gpio.c:170:9: warning: cast to smaller integer type 'uint32_t' (aka 'unsigned int') from 'uint8_t *' (aka 'unsigned char *') [-Wpointer-to-int-cast]
          if ((uint32_t)gpio_mem % PAGE_SIZE)
              ^~~~~~~~~~~~~~~~~~
      source/c_gpio.c:171:34: warning: cast to smaller integer type 'uint32_t' (aka 'unsigned int') from 'uint8_t *' (aka 'unsigned char *') [-Wpointer-to-int-cast]
              gpio_mem += PAGE_SIZE - ((uint32_t)gpio_mem % PAGE_SIZE);
                                       ^~~~~~~~~~~~~~~~~~
      2 warnings generated.
      clang -Wsign-compare -Wunreachable-code -DNDEBUG -g -fwrapv -O3 -Wall -stdlib=libc++ -I/Users/nikolaandricmitrovic/PycharmProjects/revolve2/.venv/include -I/Users/nikolaandricmitrovic/.pyenv/versions/3.11.6/include/python3.11 -c source/common.c -o build/temp.macosx-13.3-arm64-cpython-311/source/common.o
      clang -Wsign-compare -Wunreachable-code -DNDEBUG -g -fwrapv -O3 -Wall -stdlib=libc++ -I/Users/nikolaandricmitrovic/PycharmProjects/revolve2/.venv/include -I/Users/nikolaandricmitrovic/.pyenv/versions/3.11.6/include/python3.11 -c source/constants.c -o build/temp.macosx-13.3-arm64-cpython-311/source/constants.o
      clang -Wsign-compare -Wunreachable-code -DNDEBUG -g -fwrapv -O3 -Wall -stdlib=libc++ -I/Users/nikolaandricmitrovic/PycharmProjects/revolve2/.venv/include -I/Users/nikolaandricmitrovic/.pyenv/versions/3.11.6/include/python3.11 -c source/cpuinfo.c -o build/temp.macosx-13.3-arm64-cpython-311/source/cpuinfo.o
      clang -Wsign-compare -Wunreachable-code -DNDEBUG -g -fwrapv -O3 -Wall -stdlib=libc++ -I/Users/nikolaandricmitrovic/PycharmProjects/revolve2/.venv/include -I/Users/nikolaandricmitrovic/.pyenv/versions/3.11.6/include/python3.11 -c source/event_gpio.c -o build/temp.macosx-13.3-arm64-cpython-311/source/event_gpio.o
      source/event_gpio.c:24:10: fatal error: 'sys/epoll.h' file not found
      #include <sys/epoll.h>
               ^~~~~~~~~~~~~
      1 error generated.
      error: command '/usr/bin/clang' failed with exit code 1
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for RPi.GPIO
  Running setup.py clean for RPi.GPIO
  Building wheel for spidev (pyproject.toml) ... error
  error: subprocess-exited-with-error
  
  × Building wheel for spidev (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [27 lines of output]
      /private/var/folders/qf/t2v0bpyn30n1p2rqzcqj9pm40000gn/T/pip-build-env-vvxmuk5m/overlay/lib/python3.11/site-packages/setuptools/dist.py:472: SetuptoolsDeprecationWarning: Invalid dash-separated options
      !!
      
              ********************************************************************************
              Usage of dash-separated 'description-file' will not be supported in future
              versions. Please use the underscore name 'description_file' instead.
      
              By 2024-Sep-26, you need to update your project and remove deprecated calls
              or your builds will no longer be supported.
      
              See https://setuptools.pypa.io/en/latest/userguide/declarative_config.html for details.
              ********************************************************************************
      
      !!
        opt = self.warn_dash_deprecation(opt, section)
      running bdist_wheel
      running build
      running build_ext
      building 'spidev' extension
      creating build
      creating build/temp.macosx-13.3-arm64-cpython-311
      clang -Wsign-compare -Wunreachable-code -DNDEBUG -g -fwrapv -O3 -Wall -stdlib=libc++ -I/Users/nikolaandricmitrovic/PycharmProjects/revolve2/.venv/include -I/Users/nikolaandricmitrovic/.pyenv/versions/3.11.6/include/python3.11 -c spidev_module.c -o build/temp.macosx-13.3-arm64-cpython-311/spidev_module.o
      spidev_module.c:33:10: fatal error: 'linux/spi/spidev.h' file not found
      #include <linux/spi/spidev.h>
               ^~~~~~~~~~~~~~~~~~~~
      1 error generated.
      error: command '/usr/bin/clang' failed with exit code 1
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for spidev
Successfully built revolve2-ci-group revolve2-modular-robot-simulation revolve2-modular-robot revolve2-simulation revolve2-mujoco-simulator revolve2-experimentation revolve2-modular-robot-physical
Failed to build RPi.GPIO spidev
ERROR: Could not build wheels for RPi.GPIO, spidev, which is required to install pyproject.toml-based projects

surgura, 2024-01-15:
I will need more info about your system, your steps, etc.

andricmitrovic, 2024-01-15:
I followed the same steps as in the previous versions that worked and tried to install it on two different macbooks, mine has an apple processor I am not sure about the other one.

andricmitrovic, 2024-01-15:
I have 2 previous versions of revolve already installed and there was a problem with installing this one in a new environment.
Additionaly, in student student_install.sh "editable_requirements.txt" so i asummed it was "requirements_editable.txt"

andricmitrovic, 2024-01-16:
@oliverweissl Thats exactly what I need, but its not urgent i can still run hardware with revolve 1.0.0ct version

surgura, 2024-01-16:
Can you step by step post your installation procedure?