ci-group/revolve2 #515: [BUG] Incorrect limbs number in MorphologicalMeasures

andpuc23, 2024-04-11:
**Describe the bug**
MorphologicalMeasures give incorrect number of limbs for a robot

**To Reproduce**
[best_29.pickle.zip](https://github.com/ci-group/revolve2/files/14951934/best_29.pickle.zip)
remove .zip from file name, added it to put here.
```
from pyrr import Vector3

from revolve2.ci_group import terrains
from revolve2.ci_group.interactive_objects import Ball
from revolve2.ci_group.simulation_parameters import make_standard_batch_parameters
from revolve2.experimentation.logging import setup_logging
from revolve2.experimentation.rng import make_rng_time_seed
from revolve2.modular_robot import ModularRobot
from revolve2.modular_robot.body import RightAngles
from revolve2.modular_robot.body.v2 import ActiveHingeV2, BodyV2, BrickV2
from revolve2.modular_robot.brain.cpg import BrainCpgNetworkNeighborRandom
from revolve2.modular_robot_simulation import ModularRobotScene, simulate_scenes
from revolve2.simulation.scene import Pose
from revolve2.simulators.mujoco_simulator import LocalSimulator

from revolve2.ci_group.morphological_measures import MorphologicalMeasures
import pickle

def main() -> None:
    """Run the simulation."""
    # Set up logging to give output of your simulation into the command line interface (CLI).
    setup_logging()

    # Set up a random number generator, used later for the brain.
    rng = make_rng_time_seed()

    with open('best_29.pickle', 'rb') as f:
        individual = pickle.load(f)

    robot = individual.genotype.develop()

    scene = ModularRobotScene(terrain=terrains.flat())
    scene.add_robot(robot)

    simulator = LocalSimulator()

    batch_parameters = make_standard_batch_parameters()

    print(f"Number of limbs = {MorphologicalMeasures(robot.body).limbs}")

    simulate_scenes(
        simulator=simulator,
        batch_parameters=batch_parameters,
        scenes=scene,
    )

if __name__ == "__main__":
    main()
```

**Expected behavior**
Number of limbs: 4

**Actual behavior**
Number of limbs: 0

**Platform:**
 - OS: WSL + Win10, JupyterLab
 - Python Version: 3.10.12

oliverweissl, 2024-04-12:
Can you attach an image of the robot?
The implementation of limbs here only works well if there are bricks. 

I'm not sure if this is optimal but this is the design we got to include in Revolve.
If you are unsure about the workings of these measures, please ask karine miras, as she is the author of those measures.

andpuc23, 2024-04-12:
@oliverweissl sure, here you go.
[image]

Is there a list of measures which work correctly with any robot?

Thanks!

oliverweissl, 2024-04-12:
the issue in the code is that the current implementation considers only bricks, but the paper does not distinguish.
I will adapt the code to fit the paper better. If you find any other inconsistencies please let me know!#

oliverweissl, 2024-04-12:
@andpuc23 the fix is on the branch connected to this Issue, will be merged soon

closed 2024-04-16 by oliverweissl