ci-group/revolve2 #517: [BUG] Video in the wrong colorspace

kgd-al, 2024-04-12:
**Describe the bug**
Video are generated in the mujoco colorspace (BGR) while cv2 expects RGB.

**To Reproduce**
Generate any video and compare blue/red colors

**Expected behavior**
Red should be red and blue should be blue.

Replacing https://github.com/ci-group/revolve2/blob/master/simulators/mujoco_simulator/revolve2/simulators/mujoco_simulator/_simulate_scene.py#L170
with 
```
img = cv2.cvtColor(np.flipud(img), cv2.COLOR_RGBA2BGR)
```
fixes the problem on my machine (see screenshots below).

**Screenshots**
[image]
[image]
[image]

**Platform:**
 - OS: Linux 6.5.0-27-generic 28~22.04.1-Ubuntu
 - Python Version 3.10.12

oliverweissl, 2024-04-12:
will investiaget --> numpy slicing might be better to avoid dependencies.

img  = img[:,:,::-1]

kgd-al, 2024-04-12:
Isn't opencv already a dependency?
https://github.com/ci-group/revolve2/blob/master/simulators/mujoco_simulator/pyproject.toml#L34

oliverweissl, 2024-04-12:
in that case disregard my comment :D

closed 2024-04-12 by oliverweissl