ci-group/revolve2 #574: [BUG] Publishing Pypi package fail due to rename of ci-group package

oliverweissl, 2024-09-15:
Renaming package makes pypi publishing fail since no such package is defined there yet.

tckarenchiang, 2024-09-16:
It's due to the commit 13492b05936b5720e2b9cdcc994787c504922521, where we changed the package name from `ci-group` to `standards`, which is not registered on PyPI. I need to ask @surgura to give us ownership of the PyPI projects, but he is currently in the Arctic :sweat_smile:

tckarenchiang, 2024-09-16:
Also, it seems that specific job is being skipped. Not sure this is done manually or automatically?

oliverweissl, 2024-09-16:
yes you have to manually approve each release

tckarenchiang, 2024-09-17:
> yes you have to manually approve each release

Yeah, I approved that release but what I meant is, if you look at the latest commit on the master branch, we got a green check mark because the `publish` job seems to have been skipped: https://github.com/ci-group/revolve2/actions/runs/10887691891/job/30210647382. Not sure how that happened.

oliverweissl, 2024-09-17:
ahh yes because push on master doesnt publish. only generating a release triggerst the publishing pipeline

tckarenchiang, 2024-09-18:
> ahh yes because push on master doesnt publish. only generating a release triggerst the publishing pipeline

Ah I see it in the workflow now. Only a tag push can trigger the publish job. Thanks for the explanation.